nschloe/stressberry #6: Runtime Error, failed to import.

AlexCalderon02, 2017-12-07:
`RuntimeError: module compiled against API version 0xb but this version of numpy is 0xa
Traceback (most recent call last):
  File "/usr/local/bin/stressberry-plot", line 7, in <module>
    import matplotlib.pyplot as plt
  File "/usr/local/lib/python3.5/dist-packages/matplotlib/pyplot.py", line 29, in <module>
    import matplotlib.colorbar
  File "/usr/local/lib/python3.5/dist-packages/matplotlib/colorbar.py", line 32, in <module>
    import matplotlib.artist as martist
  File "/usr/local/lib/python3.5/dist-packages/matplotlib/artist.py", line 16, in <module>
    from .path import Path
  File "/usr/local/lib/python3.5/dist-packages/matplotlib/path.py", line 25, in <module>
    from . import _path, rcParams
ImportError: numpy.core.multiarray failed to import
`

Error occured when running "stressberry-plot out.dat"

nschloe, 2017-12-07:
This issue doesn't seem to be related to stressberry directly; more a matplotlib installation error. You could try and upgrade all of your pip-installed packages, e.g., with pipdate:
```
sudo -H pip install pipdate
pipdate
pipdate3
```

nschloe, 2017-12-07:
You could also attach the data file so I can try it out.

nschloe, 2018-01-30:
Closing for lack of feedback. Feel free to reopen at any time.

closed 2018-01-30 by nschloe