nschloe/stressberry #7: FileNotFoundError: [Errno 2] No such file or directory: 'stress'

gurnarok, 2018-01-30:
Installed stressberry using `pip3 install -U stressberry` and tried to run it for the first time. Got that error.

Full error message:

> Exception in thread Thread-1:
> Traceback (most recent call last):
>   File "/usr/lib/python3.5/threading.py", line 914, in _bootstrap_inner
>     self.run()
>   File "/usr/lib/python3.5/threading.py", line 862, in run
>     self._target(*self._args, **self._kwargs)
>   File "/usr/local/lib/python3.5/dist-packages/stressberry/main.py", line 54, in test_short
>     stress_cpu(4, time=300)
>   File "/usr/local/lib/python3.5/dist-packages/stressberry/main.py", line 12, in stress_cpu
>     '--timeout', '{}s'.format(time)
>   File "/usr/lib/python3.5/subprocess.py", line 266, in check_call
>     retcode = call(*popenargs, **kwargs)
>   File "/usr/lib/python3.5/subprocess.py", line 247, in call
>     with Popen(*popenargs, **kwargs) as p:
>   File "/usr/lib/python3.5/subprocess.py", line 676, in __init__
>     restore_signals, start_new_session)
>   File "/usr/lib/python3.5/subprocess.py", line 1282, in _execute_child
>     raise child_exception_type(errno_num, err_msg)
> FileNotFoundError: [Errno 2] No such file or directory: 'stress'

nschloe, 2018-01-30:
Oh yes, you'll need stress installed
```
sudo apt install stress
```

nschloe, 2018-01-30:
Fixed now.

closed 2018-01-30 by nschloe