nschloe/stressberry #9: GObject introspection?

gurnarok, 2018-01-30:
Tried to generate plot file, to see the curve and compare data.

> stressberry-plot $(date +%F)_out.dat -o $(date +%F)_stress.png
> Traceback (most recent call last):
>   File "/usr/local/lib/python3.5/dist-packages/matplotlib/backends/backend_gtk3.py", line 14, in <module>
>     gi.require_version("Gtk", "3.0")
>   File "/usr/lib/python3/dist-packages/gi/__init__.py", line 118, in require_version
>     raise ValueError('Namespace %s not available' % namespace)
> ValueError: Namespace Gtk not available
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "/usr/local/bin/stressberry-plot", line 7, in <module>
>     import matplotlib.pyplot as plt
>   File "/usr/local/lib/python3.5/dist-packages/matplotlib/pyplot.py", line 116, in <module>
>     _backend_mod, new_figure_manager, draw_if_interactive, _show = pylab_setup()
>   File "/usr/local/lib/python3.5/dist-packages/matplotlib/backends/__init__.py", line 60, in pylab_setup
>     [backend_name], 0)
>   File "/usr/local/lib/python3.5/dist-packages/matplotlib/backends/backend_gtk3agg.py", line 9, in <module>
>     from . import backend_agg, backend_gtk3
>   File "/usr/local/lib/python3.5/dist-packages/matplotlib/backends/backend_gtk3.py", line 20, in <module>
>     "Gtk3 backend requires the GObject introspection bindings for Gtk 3 "
> ImportError: Gtk3 backend requires the GObject introspection bindings for Gtk 3 to be installed.

I am running my raspberry pi in minimal mode, so no GUI or others. What packages do I need to install?

nschloe, 2018-01-30:
Actually you shouldn't need any packages. Could you attach the dat file so I can see if I reproduce this? Optionally you could also just transfer the dat file to another computer and plot the file there.

gurnarok, 2018-01-30:
Had to change the extension to TXT, since Github doesn't support DAT extension.

[2018-01-30_nofan_data.txt](https://github.com/nschloe/stressberry/files/1679085/2018-01-30_nofan_data.txt)

nschloe, 2018-01-30:
Alright, I was able to reproduce the error. Best use
```
MPLBACKEND=Agg stressberry-plot data.txt -o out.png
```
if you're converting on the raspberry. That should work.

closed 2018-01-30 by nschloe