nschloe/stressberry #10: Issues with stressberry-plot

bcutter, 2018-04-02:
Hi,

luckily I found your stresstest script, looks amazing! Well, first run is done after setting it up according to your guide.

But when it comes to creating the plot with "stressberry-plot out.dat -o out.png", I get:

```
pi@raspberry:~$ stressberry-plot out.dat -o out.png
Unable to init server: Could not connect: Connection refused
Unable to init server: Verbindung ist gescheitert:Connection refused

(stressberry-plot:12857): Gdk-CRITICAL **: gdk_cursor_new_for_display: assertion 'GDK_IS_DISPLAY (display)' failed
Traceback (most recent call last):
  File "/usr/local/bin/stressberry-plot", line 7, in <module>
    import matplotlib.pyplot as plt
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/pyplot.py", line 115, in <module>
    _backend_mod, new_figure_manager, draw_if_interactive, _show = pylab_setup()
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/backends/__init__.py", line 62, in pylab_setup
    [backend_name], 0)
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/backends/backend_gtk3agg.py", line 9, in <module>
    from . import backend_agg, backend_gtk3
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/backends/backend_gtk3.py", line 32, in <module>
    cursors.MOVE          : Gdk.Cursor.new(Gdk.CursorType.FLEUR),
TypeError: constructor returned NULL
```
Do you have any idea on where to start?

nschloe, 2018-04-02:
Are you running this on your raspberry? matplotlib doesn't have its default backend there, hence the error. Try
```
MPLBACKEND=Agg stressberry-plot out.dat [-o out.png]
```
or run on some other machine.

bcutter, 2018-04-02:
Yes I´m running it on a Raspberry Pi (3 B+). Using the command you provided doesn´t make any (positive) difference:

```
pi@raspberry:~$ MPLBACKEND=Agg stressberry-plot out.dat -o out.png
Traceback (most recent call last):
  File "/usr/local/bin/stressberry-plot", line 72, in <module>
    _main(_parse_cmd_arguments())
  File "/usr/local/bin/stressberry-plot", line 37, in _main
    plt.savefig(args.outfile, transparent=True, bbox_inches='tight')
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/pyplot.py", line 710, in savefig
    res = fig.savefig(*args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/figure.py", line 2035, in savefig
    self.canvas.print_figure(fname, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/backend_bases.py", line 2212, in print_figure
    **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/backends/backend_agg.py", line 513, in print_png
    FigureCanvasAgg.draw(self)
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/backends/backend_agg.py", line 433, in draw
    self.figure.draw(self.renderer)
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/artist.py", line 55, in draw_wrapper
    return draw(artist, renderer, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/figure.py", line 1475, in draw
    renderer, self, artists, self.suppressComposite)
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/image.py", line 141, in _draw_list_compositing_images
    a.draw(renderer)
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/artist.py", line 55, in draw_wrapper
    return draw(artist, renderer, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/axes/_base.py", line 2607, in draw
    mimage._draw_list_compositing_images(renderer, self, artists)
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/image.py", line 141, in _draw_list_compositing_images
    a.draw(renderer)
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/artist.py", line 55, in draw_wrapper
    return draw(artist, renderer, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/axis.py", line 1204, in draw
    self.label.draw(renderer)
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/artist.py", line 55, in draw_wrapper
    return draw(artist, renderer, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/text.py", line 706, in draw
    bbox, info, descent = textobj._get_layout(renderer)
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/text.py", line 305, in _get_layout
    clean_line, ismath = self.is_math_text(line, self.get_usetex())
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/text.py", line 1188, in is_math_text
    if cbook.is_math_text(s):
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/cbook/__init__.py", line 2033, in is_math_text
    "matplotlib display text must have all code points < 128 or use "
ValueError: matplotlib display text must have all code points < 128 or use Unicode strings
```
And
`pi@raspberry:~$ MPLBACKEND=Agg stressberry-plot out.dat`
doesn´t create any output.

How to uninstall and reinstall the matplotlib?

nschloe, 2018-04-02:
Oh there _is_ a difference now. What does
```
stressberry-plot -v
```
give on your system?

bcutter, 2018-04-02:
It gives

```
pi@raspberry:~$ stressberry-plot -v
Unable to init server: Could not connect: Connection refused
Unable to init server: Verbindung ist gescheitert:Connection refused

(stressberry-plot:8976): Gdk-CRITICAL **: gdk_cursor_new_for_display: assertion 'GDK_IS_DISPLAY (display)' failed
Traceback (most recent call last):
  File "/usr/local/bin/stressberry-plot", line 7, in <module>
    import matplotlib.pyplot as plt
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/pyplot.py", line 115, in <module>
    _backend_mod, new_figure_manager, draw_if_interactive, _show = pylab_setup()
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/backends/__init__.py", line 62, in pylab_setup
    [backend_name], 0)
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/backends/backend_gtk3agg.py", line 9, in <module>
    from . import backend_agg, backend_gtk3
  File "/usr/local/lib/python2.7/dist-packages/matplotlib/backends/backend_gtk3.py", line 32, in <module>
    cursors.MOVE          : Gdk.Cursor.new(Gdk.CursorType.FLEUR),
TypeError: constructor returned NULL
```

nschloe, 2018-04-02:
I meant
```
MPLBACKEND=Agg stressberry-plot -v
```
You don't have the Gtk3 backend so you'll always get these types of error messages. (So do I on my raspberry btw.)

bcutter, 2018-04-02:
That gives:

```
pi@raspberry:~$ MPLBACKEND=Agg stressberry-plot -v
stressberry 0.1.4, Python 2.7.13 (default, Nov 24 2017, 17:33:09) [GCC 6.3.0
20170516]
```
How to make progress here?

nschloe, 2018-04-02:
Alright, looks like you're using Python 2 which might cause your unicode troubles.. Try
```
MPLBACKEND=Agg python3 stressberry-plot out.dat -o out.png
```

bcutter, 2018-04-02:
Alright, I have installed:
- python3
- python3.2
- python3.2mu
- python3.5
- python3.5m
- python3m

But running "pi@RPi1raspberry:~$ MPLBACKEND=Agg python3 stressberry-plot out.dat -o out.png" for example gives:

`python3.5: can't open file 'stressberry-plot': [Errno 2] No such file or directory`

I created a virtual environment for python 3 and ran the setup command with pip again. But still when running ... i get:

```
(py3env) pi@raspberry:~$ sudo MPLBACKEND=Agg python3 stressberry-plot out.dat -o out.png
python3: can't open file 'stressberry-plot': [Errno 2] No such file or directory
```
This thing bothers me really hard. Nearly before kicking the whole stressberry thing, even it´s a very very very good software. No other users running it on a Pi 3 on Raspbian Stretch?

nschloe, 2018-04-02:
> No other users running it on a Pi 3 on Raspbian Stretch?

Well, at least myself, and everything is working fine on all machines that I've ever seen.

> This thing bothers me really hard.

I understand that. It can get really frustrating if something doesn't work straight out of the box, as admittedly stressberry should. For you it'd certainly be helpful to acquire some basic debugging skills on the command line. Here's one: If you encounter
```
python3: can't open file 'stressberry-plot': [Errno 2] No such file or directory
```
then obviously python3 hasn't found `stressberry-plot`. But it's there, right, we've just run it before? Find it via
```
which stressberry-plot
```
and then hand over the full path to python 3:
```
MPLBACKEND=Agg python3 /path/to/your/stressberry-plot out.dat -o out.png
```
Then again, since now you have python 3 installed, this
```
MPLBACKEND=Agg stressberry-plot out.dat -o out.png
```
might just work.

bcutter, 2018-04-02:
I´ve done that, but with "find" :-) So...
```

(py3env) pi@raspberry:~$ sudo MPLBACKEND=Agg python3.5 /usr/local/bin/stressberry-plot out.dat -o out.png
Traceback (most recent call last):
  File "/usr/local/bin/stressberry-plot", line 7, in <module>
    import matplotlib.pyplot as plt
ImportError: No module named 'matplotlib'
```

As I said: it feels like there´s a need for setting up stressberry once again in the python 3 environment - obviously there´s no matplotlib...

nschloe, 2018-04-02:
Well, install matplotlib. :)
```
sudo apt install python3-matplotlib
```

nschloe, 2018-04-02:
FWIW, I'd recommend you use Python3 all the time anyways. Python 2 won't be maintained for much longer and everything you'll ever need has already been ported.

bcutter, 2018-04-04:
Next after installing python3-matplotlib (and plenty other needed) packages with apt:

```
pi@raspberry:~$ MPLBACKEND=Agg python3 /usr/local/bin/stressberry-plot out.dat -o out.png
Traceback (most recent call last):
  File "/usr/local/bin/stressberry-plot", line 8, in <module>
    import yaml
ImportError: No module named 'yaml'
```

... so after an additional `sudo apt install python-yaml` and `sudo apt install python-yaml` I am VERY SURE everything needed is finally installed. Anyway I (still) get:

```
(py3env) pi@raspberry:~$ MPLBACKEND=Agg python3 /usr/local/bin/stressberry-plot out.dat -o out.png
Traceback (most recent call last):
  File "/usr/local/bin/stressberry-plot", line 7, in <module>
    import matplotlib.pyplot as plt
ImportError: No module named 'matplotlib'
```

Worst install experience ever :-D What is missing? I also installed "sudo apt install python-matplotlib && sudo apt install python3-matplotlib".

bcutter, 2018-04-04:
Hi,

I´ve set Python 3 as default now (for the specific user, not the whole system) using `alias python='/usr/bin/python3'` and `. ~/.bashrc`.

`sudo apt install python3-matplotlib` gives 
```
...
python3-matplotlib ist schon die neueste Version (2.0.0+dfsg1-2).
0 aktualisiert, 0 neu installiert, 0 zu entfernen und 4 nicht aktualisiert.
```

and `python3 -c "import matplotlib; print(matplotlib.__version__)"` gives
`2.0.0`
now. A good sign, right?

But doesn´t matter which command...
`MPLBACKEND=Agg python3 /usr/local/bin/stressberry-plot out.dat -o out.png` or
`python3 /usr/local/bin/stressberry-plot out.dat -o out.png` or
`sudo python3 /usr/local/bin/stressberry-plot out.dat -o out.png`
...they all end with the same result:
```
Traceback (most recent call last):
  File "/usr/local/bin/stressberry-plot", line 10, in <module>
    import stressberry
ImportError: No module named 'stressberry'
```

I feel like being almost there. matpoltlib seems to not be an issue anymore, it´s imported in line 7, script crashes in line 10 ("import stressberry").

bcutter, 2018-04-04:
Wait:

`sudo -H pip3 install stressberry` isn´t possible, because there´s no command "pip3". And I think pip still points to Python 2.7, right? So... how to redirect pip to the Python 3... I´ll only need to setup Stressberry there once.

nschloe, 2018-04-04:
After
```
apt install python3-pip
```
you have pip3 which installs everything for Python3. pip will still be available should you ever need to do something else in Python 2.

nschloe, 2018-04-04:
Whoohoo!! What an ordeal. Glad you got it to work, and certainly you've improved your Linux skills along the way. Congrats!

bcutter, 2018-04-04:
Yes, for python related scripts I definitely did :-)
Maybe you can add the two things which made it hard for me in the documentation or somewhere else as "requirements":
1) Use **MPLBACKEND=Agg** stressberry-plot out.dat -o out.png
2) Use Python 3.x, not < v3

nschloe, 2018-04-04:
It's all right there, isn't it?

[image]