nschloe/stressberry #15: Broken?!

Dan1jel, 2018-08-25:
I've installed both but I just get an error. So it's more then just install two things apperently.

nschloe, 2018-08-25:
```
ImportError: libf77blas.so.3: cannot open shared object file: No such file or directory
```
Seems like a broken installation.

closed 2018-08-25 by nschloe

Naphier, 2018-08-26:
Atlas needs to be installed.
`sudo apt-get install libatlas-base-dev`
But after this I get a new error. Will open a ticket.

wupasscat, 2018-09-02:
after installing atlas, this happens: 

ImportError: cairo backend requires that cairocffi or pycairo is installed
How do I fix this and why does the installation have to be so complicated!

nschloe, 2018-09-02:
@wupasscat If you have a problem, please open a new issue. Otherwise, everyone in this thread will be spammed.

> after installing atlas, this happens:

When you do what exactly?

```
ImportError: cairo backend requires that cairocffi or pycairo is installed
```
Are you running it without a graphical user interface? Then you probable require
```
MPLBACKEND=Agg stressberry-plot out.dat -o out.png
```
(See main README.)