nschloe/stressberry #16: Results for a Raspberry Pi 3B+ with FLIRC Case

RichardKav, 2018-09-25:
I thought I'd share some results, seen as I came across this project while finding a way to test the cooling of my Pi's new case.

These results are for a Raspberry Pi 3B+, using a FLIRC case, with a SanDisk Ultra Fit 128 GB USB 3.1 pen drive attached. The Pi had been left idle for a long time before the test, so the idle temperature is reasonable. The setup is pictured below:

[image]

This results in the following temperature graph, which has a peak temperature of 60.148 °C and idle temperature of 46.16 °C:

[image]

The previous case a simple acrylic case anecdotally had a resting temperature of around 52 °C, but I didn't test this before attaching the new case. The raw results for the graph can be found in the following [out.txt](https://github.com/nschloe/stressberry/files/2415325/out.txt) file.

nschloe, 2018-09-25:
Nice! I'll incorporate the results into the main readme.

nschloe, 2018-09-27:
Alright, it did go into the main readme now. Thanks again!

closed 2018-09-27 by nschloe

Szili89, 2018-11-17:
How can you output the test results in txt format as RichardKav made?
I tried:
`sudo MPLBACKEND=Agg stressberry-plot out.dat -o out.txt`
But it is not working.
`ValueError: Format 'txt' is not supported (supported formats: eps, jpeg, jpg, pdf, pgf, png, ps, raw, rgba, svg, svgz, tif, tiff)`