nschloe/stressberry #19: Unable to run on Rpi 4  4GB

flyingferret, 2019-07-03:
> pi@rasp4:~ $ MPLBACKEND=Agg stressberry-plot out.dat -o out.png
Traceback (most recent call last):
  File "/home/pi/.local/bin/stressberry-plot", line 6, in <module>
    from stressberry.cli import plot
  File "/home/pi/.local/lib/python2.7/site-packages/stressberry/__init__.py", line 15, in <module>
    from . import cli
  File "/home/pi/.local/lib/python2.7/site-packages/stressberry/cli.py", line 8, in <module>
    import matplotlib.pyplot as plt
  File "/home/pi/.local/lib/python2.7/site-packages/matplotlib/__init__.py", line 130, in <module>
    from matplotlib.rcsetup import defaultParams, validate_backend, cycler
  File "/home/pi/.local/lib/python2.7/site-packages/matplotlib/rcsetup.py", line 29, in <module>
    from matplotlib.fontconfig_pattern import parse_fontconfig_pattern
  File "/home/pi/.local/lib/python2.7/site-packages/matplotlib/fontconfig_pattern.py", line 28, in <module>
    from backports.functools_lru_cache import lru_cache
ImportError: No module named functools_lru_cache

nschloe, 2019-07-03:
Looks like an outdated matplotlib version. See https://bugs.launchpad.net/ubuntu/+source/matplotlib/+bug/1764823 and https://github.com/matplotlib/matplotlib/issues/9344. It might go away if you run it from Python 3, which should be the default anyways. I'll adapt the documentation.

nschloe, 2019-07-03:
Okay, released an update and only allow Python 3 from now. Check it out!

flyingferret, 2019-07-03:
your awesome that was a quick responce.  Thank you

nschloe, 2019-07-03:
Does it work now? Feel free to upload some data files, I'd love to include some rpi4 plots in the readme.

closed 2019-07-03 by flyingferret