nschloe/stressberry #21: Rpi 4 4gb, Naked & Active cooled Results

flyingferret, 2019-07-03:
My Setup
[image]
[RPI_4_Naked and active cooled .zip](https://github.com/nschloe/stressberry/files/3356203/RPI_4_Naked.and.active.cooled.zip)

[image]

Naked peaks at 62°C  and Active Cooled peaked at 42°C

nschloe, 2019-07-03:
Super cool! Looks like peak temperature hasn't been reached yet for the naked version. Maybe we should stress it for longer? Anyways, awesome project!

flyingferret, 2019-07-03:
My error forgot to do the runs while naked this was actually for fans OFF and fans ON

nschloe, 2019-07-04:
Alright, I included it in the main readme. Thanks again!

closed 2019-07-04 by nschloe