nschloe/stressberry #26: Unable to start Pi 4 4gb

stefanwendelmann, 2019-07-19:
I installed stressberry as said but i cant run it

Linux k4s-2 4.19.57-v7l+ #1244 SMP Thu Jul 4 18:48:07 BST 2019 armv7l GNU/Linux

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

```bash
[sudo] apt install stress
pip3 install stressberry --user
```

```bash
pi@k4s-2:~ $ sudo stressberry-run out.dat
sudo: stressberry-run: command not found
```

Flozn, 2019-07-20:
That's because the option _--user_ installs the script to ~/.local/bin which isn't in your PATH.
3 possibilities:

1. `sudo pip3 install stressberry` installs the script for all users or:
2. add ~/.local/bin to your PATH or:
3. go to ~/.local/bin and run with ./stressberry-run out.dat

I usually use option 1.

nschloe, 2019-07-21:
I usually use option 2. I put
```
PATH=$HOME/.local/bin/:$PATH
```
in my `$HOME/.bashrc`.

closed 2019-07-30 by nschloe

stefanwendelmann, 2019-08-12:
I whiped my raspbian and tired again, now i got numpy errors on the script, same with sudo :

Install steps
```
sudo pip3 install stressberry
stressberry-run out.dat
```

```
pi@raspberrypi:~ $ stressberry-run out.dat
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/dist-packages/numpy/core/__init__.py", line 17, in <module>
    from . import multiarray
  File "/usr/local/lib/python3.7/dist-packages/numpy/core/multiarray.py", line 14, in <module>
    from . import overrides
  File "/usr/local/lib/python3.7/dist-packages/numpy/core/overrides.py", line 7, in <module>
    from numpy.core._multiarray_umath import (
ImportError: libf77blas.so.3: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/bin/stressberry-run", line 6, in <module>
    from stressberry.cli import run
  File "/usr/local/lib/python3.7/dist-packages/stressberry/__init__.py", line 1, in <module>
    from . import cli
  File "/usr/local/lib/python3.7/dist-packages/stressberry/cli.py", line 7, in <module>
    import matplotlib.pyplot as plt
  File "/usr/local/lib/python3.7/dist-packages/matplotlib/__init__.py", line 138, in <module>
    from . import cbook, rcsetup
  File "/usr/local/lib/python3.7/dist-packages/matplotlib/cbook/__init__.py", line 31, in <module>
    import numpy as np
  File "/usr/local/lib/python3.7/dist-packages/numpy/__init__.py", line 142, in <module>
    from . import core
  File "/usr/local/lib/python3.7/dist-packages/numpy/core/__init__.py", line 47, in <module>
    raise ImportError(msg)
ImportError:

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy c-extensions failed.
- Try uninstalling and reinstalling numpy.
- If you have already done that, then:
  1. Check that you expected to use Python3.7 from "/usr/bin/python3",
     and that you have no directories in your PATH or PYTHONPATH that can
     interfere with the Python and numpy version "1.17.0" you're trying to use.
  2. If (1) looks fine, you can open a new issue at
     https://github.com/numpy/numpy/issues.  Please include details on:
     - how you installed Python
     - how you installed numpy
     - your operating system
     - whether or not you have multiple versions of Python installed
     - if you built from source, your compiler versions and ideally a build log

- If you're working with a numpy git repository, try `git clean -xdf`
  (removes all files not under version control) and rebuild numpy.

Note: this error has many possible causes, so please don't comment on
an existing issue about this - open a new one instead.

Original error was: libf77blas.so.3: cannot open shared object file: No such file or directory
```

And i tried to uninstall and reinstall numpy and checked the alias of python3

nschloe, 2019-08-12:
See https://github.com/nschloe/stressberry/issues/27

stefanwendelmann, 2019-08-12:
Tried it with Raspbian Desktop, turns out the raspbian light version wont work propper

stefanwendelmann, 2020-07-07:
light:
```
sudo apt install stress
sudo apt-get install libatlas-base-dev
sudo pip3 install stressberry
stressberry-run out.dat
stressberry-plot out.dat -o out.png
```