nschloe/stressberry #27: ImportError: libf77blas.so.3: cannot open shared object file: No such file or directory

sridharrajagopal, 2019-08-09:
Hi there!

I tried installing and running as you suggested, on RPi 3B+ on the latest distro. 
I got the following error:

Traceback (most recent call last):
  File "/home/pi/.local/lib/python3.5/site-packages/numpy/core/__init__.py", line 17, in <module>
    from . import multiarray
  File "/home/pi/.local/lib/python3.5/site-packages/numpy/core/multiarray.py", line 14, in <module>
    from . import overrides
  File "/home/pi/.local/lib/python3.5/site-packages/numpy/core/overrides.py", line 7, in <module>
    from numpy.core._multiarray_umath import (
ImportError: libf77blas.so.3: cannot open shared object file: No such file or directory

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./stressberry-run", line 7, in <module>
    from stressberry.cli import run
  File "/home/pi/.local/lib/python3.5/site-packages/stressberry/__init__.py", line 1, in <module>
    from . import cli
  File "/home/pi/.local/lib/python3.5/site-packages/stressberry/cli.py", line 7, in <module>
    import matplotlib.pyplot as plt
  File "/home/pi/.local/lib/python3.5/site-packages/matplotlib/__init__.py", line 141, in <module>
    from . import cbook, rcsetup
  File "/home/pi/.local/lib/python3.5/site-packages/matplotlib/cbook/__init__.py", line 33, in <module>
    import numpy as np
  File "/home/pi/.local/lib/python3.5/site-packages/numpy/__init__.py", line 142, in <module>
    from . import core
  File "/home/pi/.local/lib/python3.5/site-packages/numpy/core/__init__.py", line 47, in <module>
    raise ImportError(msg)
ImportError: 

IMPORTANT: PLEASE READ THIS FOR ADVICE ON HOW TO SOLVE THIS ISSUE!

Importing the numpy c-extensions failed.
- Try uninstalling and reinstalling numpy.
- If you have already done that, then:
  1. Check that you expected to use Python3.5 from "/usr/bin/python3",
     and that you have no directories in your PATH or PYTHONPATH that can
     interfere with the Python and numpy version "1.17.0" you're trying to use.
  2. If (1) looks fine, you can open a new issue at
     https://github.com/numpy/numpy/issues.  Please include details on:
     - how you installed Python
     - how you installed numpy
     - your operating system
     - whether or not you have multiple versions of Python installed
     - if you built from source, your compiler versions and ideally a build log

- If you're working with a numpy git repository, try `git clean -xdf`
  (removes all files not under version control) and rebuild numpy.

Note: this error has many possible causes, so please don't comment on
an existing issue about this - open a new one instead.

Original error was: libf77blas.so.3: cannot open shared object file: No such file or directory

On searching further, I found this thread - https://github.com/Kitt-AI/snowboy/issues/262

where the solution was to install 
sudo apt-get install libatlas-base-dev

I did that and was able to run stressberry-run successfully,. 

I don't know if you need to update your instructions, or if it is a peculiarity with my setup. 

Thanks,
Sridhar

nschloe, 2019-08-09:
> I don't know if you need to update your instructions, or if it is a peculiarity with my setup.

Seems like a borked system installation. When installing stress via `sudo apt install stress`, all dependencies should be pulled in. No idea why this wasn't the case on your machine.

closed 2019-08-09 by nschloe

AlmightyFrog, 2019-11-17:
I've had the same issue - what about adding a short note in the readme about it?

nschloe, 2019-11-17:
This is a broken installation, so not necessarily in the domain of stressberry. I might add something to the readme if more people run into this though.

AlmightyFrog, 2019-11-17:
Not sure where that broken installation would have came from. It's a pretty clean RPI4 Raspbian Buster Lite installation.

So if i got your comment above correctly you tell this is a dependency of stress. Could you please paste what it says on your raspberry?
Here it is:
> $ apt depends stress
> stress
>   Depends: libc6 (>= 2.7)

There is a nice script at a gist (https://gist.githubusercontent.com/sr105/09d2534d467fcd8b881e/raw/713ca8f0977efc72b6d18f91f1702c3200d95dc4/apt-rdepends-tree) which draws you a full tree with all dependencies. Here you go, also there I can't find libatlas-base-dev as dependency:

> $ ./apt-rdepends-tree.sh stress
> Reading package lists... Done
> Building dependency tree       
> Reading state information... Done
> 
>     ├─ stress
>     │    └─ libc6 (>= 2.7)
>     ├─ libc6
>     │    └─ libgcc1
>     ├─ libgcc1
>     │    ├─ gcc-9-base (= 9.2.1-19)
>     │    └─ libc6 (>= 2.4)
>     └─ gcc-9-base

Verifying reverse for libatlas-base-dev I also can't find it:
> $ apt-rdepends -r libatlas-base-dev | grep stress
> Reading package lists... Done
> Building dependency tree       
> Reading state information... Done

Would be nice if you could have a look on your side, as I don't have any RPI running with the full image.

nschloe, 2019-11-17:
You're right, it's not from stress then. Any idea where else it could come from?

AlmightyFrog, 2019-11-17:
Not sure, but here is full output from apt-rdepends -r libatlas-base-dev, maybe you can find the application you installed there:

> libatlas-base-dev
>   Reverse Depends: libhypre-dev (2.15.1-5)
>   Reverse Depends: libjulia1 (1.0.3+dfsg-4+rpi1)
>   Reverse Depends: libmathlib2-dev (20061220+dfsg3-4.4)
>   Reverse Depends: libsuperlu-dev (5.2.1+dfsg1-4)
>   Reverse Depends: mapdamage (2.0.9+dfsg-1)
>   Reverse Depends: odin (2.0.3-1)
>   Reverse Depends: r-base-dev (3.5.2-1)
>   Reverse Depends: xmds2 (2.2.3+dfsg-15)
> libhypre-dev
>   Reverse Depends: libpetsc-real3.10-dev (>= 3.10.3+dfsg1-5)
>   Reverse Depends: libpetsc-real3.11-dev (>= 3.11.4+dfsg1-3)
>   Reverse Depends: libsundials-dev (3.1.2+dfsg-3+b1)
> libpetsc-real3.10-dev
>   Reverse Depends: libdolfin-dev (2018.1.0.post1-16)
>   Reverse Depends: libpetsc-real-dev (3.10.3+dfsg1-5)
>   Reverse Depends: libpetsc-real3.10-dbg (= 3.10.3+dfsg1-5)
>   Reverse Depends: libslepc-real3.10-dev (3.10.1+dfsg1-3)
> libdolfin-dev
>   Reverse Depends: dolfin-doc (>= 2018.1.0.post1-16)
>   Reverse Depends: fenics (>= 1:2018.1.0.6+b19)
>   Reverse Depends: libmshr-dev (>= 2018.1.0+dfsg1-7+b2)
>   Reverse Depends: python3-dolfin (= 2018.1.0.post1-16)
> dolfin-doc
>   Reverse Depends: fenics (>= 1:2018.1.0.6+b19)
> fenics
> libmshr-dev
>   Reverse Depends: python3-mshr (= 2018.1.0+dfsg1-7+b2)
> python3-mshr
>   Reverse Depends: fenics (>= 1:2019.1.0.3)
> python3-dolfin
>   Reverse Depends: dolfin-bin (>= 2018.1.0.post1-16)
>   Reverse Depends: dolfin-doc (>= 2018.1.0.post1-16)
>   Reverse Depends: fenics (>= 1:2018.1.0.6+b19)
>   Reverse Depends: python-dolfin (>= 2018.1.0.post1-16)
>   Reverse Depends: python3-mshr (>= 2018.1.0+dfsg1-7+b2)
> dolfin-bin
>   Reverse Depends: fenics (>= 1:2018.1.0.6+b19)
>   Reverse Depends: python3-mshr (>= 2018.1.0+dfsg1-7+b2)
> python-dolfin
> libpetsc-real-dev
>   Reverse Depends: petsc-dev (3.11.4+dfsg1-3)
> petsc-dev
>   Reverse Depends: libsundials-dev (3.1.2+dfsg-3+b1)
> libsundials-dev
> libpetsc-real3.10-dbg
> libslepc-real3.10-dev
>   Reverse Depends: libdolfin-dev (2018.1.0.post1-16)
>   Reverse Depends: slepc-dev (3.10.1+dfsg1-3)
> slepc-dev
> libpetsc-real3.11-dev
>   Reverse Depends: libdolfin-dev (2019.1.0-8)
>   Reverse Depends: libpetsc-real-dev (3.11.4+dfsg1-3)
>   Reverse Depends: libpetsc-real3.11-dbg (= 3.11.4+dfsg1-3)
>   Reverse Depends: libslepc-real3.11-dev (3.11.2+dfsg1-1+b1)
> libpetsc-real3.11-dbg
> libslepc-real3.11-dev
>   Reverse Depends: libdolfin-dev (2019.1.0-8)
>   Reverse Depends: slepc-dev (3.11.2+dfsg1-1)
> libjulia1
>   Reverse Depends: julia (= 1.0.4+dfsg-1)
>   Reverse Depends: libjulia-dev (= 1.0.3+dfsg-4+rpi1)
> julia
>   Reverse Depends: julia-factcheck (0.1.2-2)
> julia-factcheck
> libjulia-dev
> libmathlib2-dev
>   Reverse Depends: cernlib-core-dev (20061220+dfsg3-4.4)
[... 86 lines cut ...]
> libdune-istl-dev
>   Reverse Depends: libdune-pdelab-dev (>= 2.6~20180302-2)
> libdune-pdelab-dev
> libgetfem++-dev
> libpetsc-complex3.10-dev
>   Reverse Depends: libpetsc-complex-dev (3.10.3+dfsg1-5)
>   Reverse Depends: libpetsc-complex3.10-dbg (= 3.10.3+dfsg1-5)
>   Reverse Depends: libslepc-complex3.10-dev (3.10.1+dfsg1-3)
> libpetsc-complex-dev
> libpetsc-complex3.10-dbg
> libslepc-complex3.10-dev
> libpetsc-complex3.11-dev
>   Reverse Depends: libpetsc-complex-dev (3.11.4+dfsg1-3)
>   Reverse Depends: libpetsc-complex3.11-dbg (= 3.11.4+dfsg1-3)
>   Reverse Depends: libslepc-complex3.11-dev (3.11.2+dfsg1-1+b1)
> libpetsc-complex3.11-dbg
> libslepc-complex3.11-dev
> mapdamage
> odin
> r-base-dev
>   Reverse Depends: dh-r (20190121)
>   Reverse Depends: r-base-core-dbg (= 3.5.2-1)
> dh-r
> r-base-core-dbg
> xmds2

rharmonson, 2020-01-30:
Experienced the same issue and symptoms, googled, found this issue, and installed libatlas-base-dev which corrected the problem.

I am using a default installation with only the addition of stress, pip3, stressberry, python3-numpy, and now libatlas-base-dev. Also, apt update/upgrade this evening, January 29, 2020.

Raspbian Buster Lite
Version: September 2019
Release date: 2019-09-26