nschloe/stressberry #31: Running stressberry on RPi4 in a solid KKSB milled case

JohBod, 2019-08-27:
My new KKSB case soaked up the CPU heat pretty well and the large case area forwarded the heat into the air through convection. No fan in sight, it appears to stay near 60 degrees C, no clock throttling detected.

[image]

[image]

nschloe, 2019-08-31:
What a nice case! Could you also attach the data file so I can merge it into the figure on the README?

JohBod, 2019-09-10:
I replied to Github's email notification about your message above about a week ago. I got no response so, in case you didn't get it, i'll attach the file here as well. The old graph above is from running the test with the case partially covered with a paper. The attached file is from running the test with the case standing free and alone on the table to allow free air circulation.

[out.txt](https://github.com/nschloe/stressberry/files/3596822/out.txt)

nschloe, 2019-09-10:
Cool, I'll add this to the readme.

nschloe, 2019-09-11:
Alright, added. Thanks again!

closed 2019-09-11 by nschloe