nschloe/stressberry #37: Reporting results on an overclocked Raspberry Pi 4B with Argon One case

jholloway, 2019-09-21:
Got a RPi4 a few weeks ago and have been tweaking and setting it up ever since. I had my eye on the [Argon One](https://www.argon40.com/argon1/argon-one-pi4.html) case for it, and it just arrived today.

The case has an aluminum top plate that directly connects to the chips on the board, similar to the well-known [Flirc case](https://flirc.tv/more/raspberry-pi-4-case). It also has an active fan in it.

This board will be used as a retro gaming setup with RetroArch, and I wanted to overclock it to eke out the maximum performance in early 3D platforms like PlayStation, N64, and Dreamcast. The Pi CPU is overclocked to 1950 MHz. (The GPU and 3D blocks are overclocked as well, but I don't think it matters for this test.)

Anyway, I'm very pleased with the results. Even with an aggressive overclock, it keeps things very cool!

[image]

Results file, as TXT since GitHub won't allow me to upload it directly as DAT: [out2.txt](https://github.com/nschloe/stressberry/files/3638334/out2.txt)

nschloe, 2019-09-25:
Thanks, added this to the main readme.

closed 2019-09-25 by nschloe

willtejeda, 2019-12-02:
I just placed the order for mine, I'm curious how yours has held up 

Have you tried the new 2.15ghz firmware ?

dennbenn73, 2020-01-17:
Have you had any issues of the daughter board consuming so much power that it limits and/or prevents the 2.1ghz overclock? I would like to know before I buy this case.