nschloe/stressberry #42: AttributeError: 'NoneType' object has no attribute 'plot'

voltlog, 2019-09-27:
I'm having trouble plotting data while running on the raspberry pi 4 itself, command line. (Raspbian 10 buster).

```
 ./stressberry-plot out.dat -o out.png
Unable to init server: Could not connect: Connection refused
Unable to init server: Could not connect: Connection refused

(stressberry-plot:9112): Gdk-CRITICAL **: 10:13:18.390: gdk_cursor_new_for_display: assertion 'GDK_IS_DISPLAY (display)' failed

(stressberry-plot:9112): Gdk-CRITICAL **: 10:13:18.393: gdk_cursor_new_for_display: assertion 'GDK_IS_DISPLAY (display)' failed
Traceback (most recent call last):
  File "./stressberry-plot", line 10, in <module>
    sys.exit(plot())
  File "/home/pi/.local/lib/python3.7/site-packages/stressberry/cli.py", line 152, in plot
    ax1.plot(
AttributeError: 'NoneType' object has no attribute 'plot'
```
Any idea what I might be wrong here?

nschloe, 2019-09-30:
> Any idea what might be wrong here?

Yes. Main readme:
> (Use MPLBACKEND=Agg stressberry-plot out.dat -o out.png if you're running the script on the Raspberry Pi itself.)

voltlog, 2019-10-01:
I forgot to mention in the post above that I've tried the command mentioned in the readme, I get the same error related to attribute 'plot':
` MPLBACKEND=Agg stressberry-plot out.dat -o out.png
Traceback (most recent call last):
  File "/home/pi/.local/bin/stressberry-plot", line 10, in <module>
    sys.exit(plot())
  File "/home/pi/.local/lib/python3.7/site-packages/stressberry/cli.py", line 152, in plot
    ax1.plot(
AttributeError: 'NoneType' object has no attribute 'plot'
`

nschloe, 2019-10-01:
Yeah, that is really weird, never seen this before. I cannot reproduce it either, so you'll have to do some debugging. The line
```
ax1 = fig.add_subplot()
```
should return an axis object, not `None`.

spiff72, 2019-10-02:
I am getting the same error here:

`pi@raspberrypi:~ $ MPLBACKEND=Agg stressberry-plot out.dat -o out.png
Traceback (most recent call last):
  File "/home/pi/.local/bin/stressberry-plot", line 10, in <module>
    sys.exit(plot())
  File "/home/pi/.local/lib/python3.7/site-packages/stressberry/cli.py", line 152, in plot
    ax1.plot(
AttributeError: 'NoneType' object has no attribute 'plot'
`

I am running on a RasPi 4 with 4GB RAM, with the latest Raspbian updates applied.

spiff72, 2019-10-02:
I should add that the stressberry-run portion runs without issue.  just the plot portion fails.

spiff72, 2019-10-04:
Can this issue be re-opened?  I just checked this on a raspberry pi zero running buster, and get the same issue.

I also installed stressberry on a different pi zero running raspbian strech and the same error happens there too.

nschloe, 2019-10-04:
>  that is really weird, never seen this before. I cannot reproduce it either, so you'll have to do some debugging.

voltlog, 2019-10-04:
> > that is really weird, never seen this before. I cannot reproduce it either, so you'll have to do some debugging.

Can you confirm that you've tried the latest raspbian release, followed the readme instructions alone and it works? (I've tried on a raspberry pi 4 and the plot function does not work, giving the error mentioned above)

spiff72, 2019-10-04:
> Can you confirm that you've tried the latest raspbian release, followed the readme instructions alone and it works? (I've tried on a raspberry pi 4 and the plot function does not work, giving the [...]

I would be curious about this too.  If nschloe was to start from the current raspbian  release image (buster) on a fresh install, do the instructions above work?  I am thinking that there may be something that the dev has set up that may have changed with some recent updates (even updates to stretch - not just buster).

When I set this up last night on my Pi Zero with Stretch installed, I ran into a few more glitches in the process (namely the numpy issue mentioned elsewhere in this [closed thread](https://github.com/nschloe/stressberry/issues/27)).  I solved this by following the recommendation in that thread to install:

`sudo apt-get install libatlas-base-dev`

I didn't run into the numpy issue when installing on buster (either on Raspi 4 or Raspi Zero - both clean installs).  I only had to fix my path so the .local/bin and local/bin were included.

lagurus, 2019-10-04:
For me help,

`pip3 install -U matplotlib`

spiff72, 2019-10-04:
> For me help,
> 
> `pip3 install -U matplotlib`

Oh snap - that worked!  Thanks lagurus!  What does the -U option do in this case?

moose4lord, 2019-10-05:
I was having the "AttributeError: 'NoneType' object has no attribute 'plot'" error on a fresh install of Raspbian on my new Pi 4. I ran the "pip3 install -U matplotlib" command (thanks @lagurus), which helped, but I still got this error:
```
pi@raspberrypi4:~ $ MPLBACKEND=Agg stressberry-plot stressberry.dat
/usr/local/lib/python3.7/dist-packages/stressberry/cli.py:197: UserWarning: Matplotlib is currently using agg, which is a non-GUI backend, so cannot show the figure.
  plt.show()
```
I'm a noob, so don't know what that means, but running the command "stressberry-plot stressberry.dat" (without the "MPLBACKEND=Agg" prefix) seems to work.

nschloe, 2019-10-05:
@moose4lord That's not an error, it's a just warning (that you won't see a window with the plot, as you would expect on a desktop.)

closed 2019-10-05 by nschloe

voltlog, 2019-10-09:
> `pip3 install -U matplotlib`

I can confirm this fixes the problem for me too. Thanks!

noderat, 2019-12-26:
@nschloe 
> Yeah, that is really weird, never seen this before. I cannot reproduce it either, so you'll have to do some debugging. The line
> 
> ```
> ax1 = fig.add_subplot()
> ```
> 
> should return an axis object, not `None`.

A freshly installed and updated Raspbian install running `pip3 install --user stressberry` will only install matplotlib 3.0.2. Updating works because that feature was added in matplotlib 3.1.0+

https://matplotlib.org/3.1.0/api/api_changes.html#figure-add-subplot-with-no-arguments

> Calling Figure.add_subplot() with no positional arguments used to do nothing; this now is equivalent to calling add_subplot(111) instead.