nschloe/stressberry #43: No such file or directory: 'out.dat' + other messages

AverageMaker, 2019-10-15:
When I installed as per the guidance (on a Pi4 with Buster full version), this message came up in yellow:
```
The scripts stressberry-plot and stressberry-run are installed in '/home/pi/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
```
I don't really know what that wants me to do, so I continued. Now when I try the command, I get:
`-bash: stressberry-plot: command not found`

I find that stressberry is installed to:
`/home/pi/.local/bin`

So I tried running it from there and still no joy.  I tried installing with a different command (`sudo -H pip3 install stressberry`) and got a bit further, but still not running:
`stressberry-plot: error: argument infiles: can't open 'out.dat': [Errno 2] No such file or directory: 'out.dat'`

I then try creating an empty out.dat file and tried again. Now I get:
```
pi@raspberrypi:~ $ sudo MPLBACKEND=Agg stressberry-plot out.dat
Traceback (most recent call last):
  File "/usr/local/bin/stressberry-plot", line 10, in <module>
    sys.exit(plot())
  File "/usr/local/lib/python3.7/dist-packages/stressberry/cli.py", line 146, in plot
    terminal_temps = [d["temperature"][-1] for d in data]
  File "/usr/local/lib/python3.7/dist-packages/stressberry/cli.py", line 146, in <listcomp>
    terminal_temps = [d["temperature"][-1] for d in data]
TypeError: 'NoneType' object is not subscriptable
```

I've gone as far as my limited Linux abilities will allow me to get, so I'm here asking for help. The instructions seem very clear and straight forward, but it just won't seem to work for me.

I was initially using Raspbian Lite but changed to the full version in case some kind of dependency is missing - but still no joy. FYI I'm using SSH from my PC to do this.

nschloe, 2019-10-16:
Okay, first some command-line basics: If you type anything in your command line, e.g., `ls`, Linux searches for a program called `ls` in certain directories, e.g., `/usr/bin/`, `/usr/local/bin/` etc. You see all those directories when looking at the `$PATH` environment variable:
```
echo $PATH
```
The different directories are separated by `:`. You can change that path by typing
```
export PATH=/usr/bin:/usr/sbin
```
or extend it by
```
export PATH=$PATH:/my/directory/
```
The message
```
The scripts stressberry-plot and stressberry-run are installed in '/home/pi/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
```
suggests just that. Note that you'll have to `export` the new path every time you log in. Because this can get fairly annoying, many people put
```
export PATH=$PATH:/home/pi/.local/bin
```
in their `.bashrc` file -- a file that gets executed every time you log in.
---

Before you can plot anything, note that you'll first have to _run_ the stress test. As the main readme says:
```
stressberry-run out.dat
stressberry-plot out.dat -o out.png
```

nschloe, 2019-11-01:
Closing for lack of feedback. Feel free to reopen at any time.

closed 2019-11-01 by nschloe