nschloe/stressberry #47: CooliPi 4B with fan on

CooliPi, 2019-11-17:
[image]
[image]

[CooliPi-fan_on.zip](https://github.com/nschloe/stressberry/files/3856137/CooliPi-fan_on.zip)

Hi,

the length of the test (5 minutes ?) is way too short. It takes half an hour for the temperature to settle. See https://www.coolipi.com/Performance.html
I also miss ambient temperature, because you can't make a fair comparison if you don't know it.

Mine was 25.5-26 degC. Raspberry Pi 4  model B. Overclocked to 1850MHz, over_voltage=2
GPU wasn't intentionally overclocked. Monitor - 1920x1200 via HDMI.

The best stress test isn't stress or stress-ng, but linpack. At least for Raspberry Pi 4.

I've also noticed different cooling performance with a monitor connected and then without a monitor. 4K monitor would make the device heat up even more.

So I suggest you also require these metadata.

CooliPi 4B is primarily a passive heatsink with a case, this setup is mainly for industrial users and for overclockers.

Best Regards.

Lada (lada@coolipi.com)

CooliPi, 2019-12-01:
I've also found out differences in temperature with different RPI firmwares. The latest one slashes some more degrees of temperature. So I suggest you also collect the firmware version. Not sure how many Celsius degrees of temperature can it slash yet with a fan on. This submitted performance log was collected using the a previous firmware (than the latest).

nschloe, 2019-12-03:
Thanks added those to the readme.

Added an issue for the firmware version. https://github.com/nschloe/stressberry/issues/52

> the length of the test (5 minutes ?) is way too short. 

It's good enough for the comparison between cases. I might extend it in the future though.

> I also miss ambient temperature, because you can't make a fair comparison if you don't know it.

The raspberry doesn't have a sensor for the ambient temp which is why I didn't include it. You're right that this isn't very scientific.

> I've also noticed different cooling performance with a monitor connected and then without a monitor. 4K monitor would make the device heat up even more.

Interesting!

closed 2019-12-03 by nschloe

shortbloke, 2019-12-03:
@CooliPi My fork of this project includes using capturing the Ambient temperature using a DHT11, DHT22 or AM2302 based sensor. I think this is necessary if the goal is to compare cases with the fewest variables impacting the comparisons. https://github.com/shortbloke/stressberry 

In addition, the idle, and run stress run times are now configurable via the command line. 

I accept this isn't all that useful when looking for results linked in the readme here, since they are intentionally all running exactly the same test, but in unknown conditions.  I published my own results on my blog https://martinrowan.co.uk where I standardised on a 1 hour test. 

There are many variables that will impact the temperature: 
* GUI vs Terminal
* Background processes
* Attached USB devices
* WiFi or Ethernet
* Firmware versions of specific components
* Screen resolution
* Enabling 4k display support
* Overclocking
+ more

PR #51 (2019-12-03): add coolipi images
Fixes #47, fixes #48.