nschloe/stressberry #48: CooliPi 4B in passive mode

CooliPi, 2019-11-17:
[image]
[image]
[CooliPi-passive.zip](https://github.com/nschloe/stressberry/files/3856170/CooliPi-passive.zip)
[image]

Hi,

the length of the test (5 minutes ?) is way too short. It takes half an hour for the temperature to settle. See https://www.coolipi.com/Performance.html
I also miss ambient temperature, because you can't make a fair comparison if you don't know it.

Mine was 25.5-26 degC. Raspberry Pi 4  model B. Overclocked to 1850MHz, over_voltage=2
GPU wasn't intentionally overclocked. Monitor - 1920x1200 via HDMI.

The best stress test isn't stress or stress-ng, but linpack. At least for Raspberry Pi 4.

I've also noticed different cooling performance with a monitor connected and then without a monitor. 4K monitor would make the device heat up even more.

So I suggest you also require these metadata.

CooliPi 4B is primarily a passive heatsink with a case, this setup is mainly for industrial users and for overclockers.

Best Regards.

Lada (lada@coolipi.com)

CooliPi, 2019-12-01:
I've also found out differences in temperature with different RPI firmwares. The latest one slashes some more degrees of temperature. So I suggest you also collect the firmware version. It can slash 2 degrees Celsius at idle. This submitted performance log was collected using the a previous firmware (than the latest).

nschloe, 2019-12-03:
> The best stress test isn't stress or stress-ng, but linpack. At least for Raspberry Pi 4.

How?

closed 2019-12-03 by nschloe

CooliPi, 2020-01-13:
V Tue, 03 Dec 2019 04:55:20 -0800
Nico Schlömer <notifications@github.com> napsáno:

> > The best stress test isn't stress or stress-ng, but linpack. At
> > least for Raspberry Pi 4.  
> 
> How?
> 

Hello Nico,

sorry for a month delay - I was super busy. I've attached a linpack
benchmark compiled for 32bit raspbian. It can be compiled for a 64bit
system too, just some optimisations to gcc need to be different. It
uses Neon in 64bit mode, hopefully in 32bit too.

I've made two scripts, sequence.sh and run.sh  They're straighforward.
I'm logging temperature sideways by logujteplotu.sh script. I use a
different log format myself, one that has one line per sample and
columns are linear time, temperature and CPU frequency. Gnuplotting
is easier then. At least for myself. The linear absolute time can be
normalised to some start value by filtering it with a script.

You can see a chart plot by gnuplot with linpack as load. I use stepped
load - i.e. adding threads after each half an hour, because it can very
well illustrate what's going on.

See https://www.coolipi.com/Performance.html

The plots' coloring is automatic, based on a third column - cpu
frequency. Some gnuplot conditional chicanery is involved, I can dig it
out and send if requested. It's more clear than using a second plot
with frequencies, imho.

I've even tried to run glxgears alongside 4 linpack threads, but it
decreased temperature by about 1-2°C because it has stolen cpu cycles
from heavy load.

I don't know which github repo was used for the linpack benchmark, but
it should be very easy to find the sources online. I'm sending it as I
got it from my friend. If you don't trust the sources or a binary, find
it yourself and compile. Compile command is somewhere in a README -
also easy to find.

Have fun playing with it and best regards,

Lada Myslik

PR #51 (2019-12-03): add coolipi images
Fixes #47, fixes #48.