nschloe/stressberry #50: output .png file and got error

bigbear1024, 2019-12-03:
Hi, I trying to output the png file, and show error message like below, what should I do?
Thanks!

> pi@raspberrypi:~/Desktop $ MPLBACKEND=Agg stressberry-plot out.dat -o out.png
Traceback (most recent call last):
  File "/home/pi/.local/bin/stressberry-plot", line 8, in <module>
    sys.exit(plot())
  File "/home/pi/.local/lib/python3.7/site-packages/stressberry/cli.py", line 152, in plot
    ax1.plot(
AttributeError: 'NoneType' object has no attribute 'plot'

closed 2019-12-03 by bigbear1024

anujdeshpande, 2020-01-14:
@bigbear1024 how did you fix this?

bigbear1024, 2020-01-14:
@anujdeshpande you can check this issue
https://github.com/nschloe/stressberry/issues/42#issue-499334260