nschloe/stressberry #53: ProgPi passiv cooling case

progpi, 2019-12-17:
My temperature measurements on Pi 4 are done with some scripts based on sysbench + stressberry-plot Picture from my Pi 4 cooling solution : 
[https://progpi.de/cpu-stresstestverfahren/](url)

Temperatures under 50°C after 60 min of stresstest.

nschloe, 2019-12-28:
If you'd like me to include the data into the main readme, you'll actually have to post the data here.

progpi, 2019-12-28:
A cylindric piece of copper is above the Pi 4 SoC and combined with heatsinks for LED's
[image]

The used case is StackBT (Reichelt) but has many different names. My temperature measurements on Pi 4 are done with some scripts based on sysbench + stressberry-plot 
 
[image]

Done with 60 min of sysbench (4 core Test).  Due to the specific heat capacity of copper a shorter test time doesn't give representative temperatures. More info here : 
https://progpi.de/cpu-stresstestverfahren/

nschloe, 2019-12-30:
All very nice, thanks for sharing your experiments here! I'll close the report since there nothing actionable for stressberry now, but feel free to add more information for others to find.

progpi, 2020-01-01:
Will the ProgPi-Case (with copper/wood cover plate) become visible on "main readme" ?

nschloe, 2020-01-02:
That's only possible of you post the data here.

progpi, 2020-01-02:
my data files are for a cycle of 10+60+10 Minutes with Temperature measurement in 10 sec steps 
[pi4cpu.zip](https://github.com/nschloe/stressberry/files/4014693/pi4cpu.zip)

nschloe, 2020-01-02:
Okay, this is different from all other data in the readme so it'll be difficult to include those. Why not produce 5+10+5 files like all the others?

progpi, 2020-01-02:
Well I could do so, but you see the temperature increasing a lot due to the capacity of copper plate, copper cylinder and the LED heatsinks. But I think it would be sufficient just to have a Link to this post like it is done with CooliPi

nschloe, 2020-01-02:
The first diagram in the readme gives some ballpark figures, and for that it works quite alright. If you want to compare some heavy solutions where you need 1 degree precision, then that's best done on a separate page as you did.

progpi, 2020-01-02:
Meanwhile I shortend my measurements to 15 min giving the following results
[image]
ProgPi-Case is missing here because I have only one Pi4 with 2GB. Be free to test sberry-run15.sh
with your own cases. In the png there are some added *.dat coming from "IceTower"+CooliPi.
 sberry-run15.sh adds new measurements to *.cfg and starts then plotrun15.sh - easy to use I think.

nschloe, 2020-01-02:
Okay, I've added some plots to the main readme now.

closed 2020-01-02 by nschloe