nschloe/stressberry #54: Ice Tower Low Profile Cooler - Test Results

leonhess, 2019-12-17:
Hey there,
I did a bunch of tests using the Ice Tower Low Profile version Cooler.
[IceTowerResults.zip](https://github.com/nschloe/stressberry/files/3975362/IceTowerResults.zip)

[image]
[image]

I did tests without the Fan, with the Fan and also swapped the included Fan for a Noctua Fan. I also tested the Overclocking capabilities of this cooler.

For the Overclocks I used these settings:

| Frequency  | overvolt level |
| ------------- | ------------- |
| 1500 | 0 |
| 1750 | 2 |
| 2000 | 4 |
| 2147 | 6 |

Firmware:
```
BOOTLOADER: up-to-date
CURRENT: Tue 10 Sep 10:41:50 UTC 2019 (1568112110)
 LATEST: Tue 10 Sep 10:41:50 UTC 2019 (1568112110)
VL805: up-to-date
CURRENT: 000137ab
 LATEST: 000137ab
```

[image]

First of looking at the passive capabilities, you can see that Even a hefty overclock to 2147MHz is doable, though I am not quite sure if it has stabilised in 5 minutes.
[image]

Then comparing the 2 used fans one can see that the Noctua is about 1-2 °C cooler but I think it also spins faster.
[image]
[image]

Then here is a graph with all the results with the Noctua fan:
[image]

And lastly I swapped the included thermal pad for some thermal grease which especially gave a big boost in performance for the 2.147GHz overclock.
[image]
[image]

I am not sure how much of this you would want to include in README?

nschloe, 2019-12-28:
Thanks, I've included some data in the main readme. Thanks in particular for the stock data, this had been sorely missing for a while!

closed 2019-12-28 by nschloe

AmirulAndalib, 2023-07-06:
[image]

Overvoltage 10

2321mhz cpu, 770mhz gpu

PR #55 (2019-12-28): add ice tower
Fixes #54.