nschloe/stressberry #57: Raspberry Pi 4B@1800MHz / Joy-It alu case with PiFan@3V3

Luzypher, 2020-01-09:
I submit you my own data for displaying this on your project page. It's about a Raspberry Pi 4B 4GB with actual firmware/eeprom in a "Joy-It" aluminium case. On the top side are small holes for airflow, but without a fan, the temperature raises up to over 80°C in stressberry test!
I add a small (30x30 mm) fan 5V (also usable with 3.3V) under the top cover for blowing cool air into the case and over the Raspberry Pi board (and ARM cpu). I use the 3V3 and ground pins from the Pi for a low noise operating, because the small fan is too loud at 5V!
With default clocking (arm/gpu@500/1500MHz) the temperatures raises only up to ~53°C with stressberry-test. At arm/gpu@600/1800MHz with a little but noticeable performance boost the temperatures raises up to ~64°C. Not bad at all for a low cost solution :)

Some other data:
- original Raspberry Pi power supply 5.1V/3A
- Raspbian Buster with 64bit kernel installed
- Joy-It aluminium case for round about 12 €
- "PiFan" 30x30 mm fan 5V (also for 3.3V) for round about 3 €

[out1800.dat.zip](https://github.com/nschloe/stressberry/files/4041790/out1800.dat.zip)
[image]
[image]

cu ...
Dirk

Luzypher, 2020-01-09:
... and the output png file.

[image]

progpi, 2020-01-12:
Hi Dirk, I used a very simple "StackBT-Case" in a quite similar configuration (Pi-Fan blowing on the SoC) and I got 50°C with 16Ohm resistor between and connected to 5V - avoiding 3.3V it's my low noise mode...  In #53 (closed issue in the moment) I used the following JoyIT-AluCase : 
[image]

This case is for passive use and has cooling ribs on top/bottom.  Some temperature diagrams are on https://progpi.de/cpu-stresstestverfahren/ and #53. So I suggest not to write "JoyIt-AluCase" only...
Regards Harald