nschloe/stressberry #60: Raspberry Pi 4B [1.5GHz vs 2.15GHz]

haykh, 2020-01-22:
I'm using RPi 4B (with 4GB memory) embedded in the [Smraza case](https://www.amazon.com/gp/product/B07X8RL8SL/ref=ppx_yo_dt_b_asin_title_o01_s00?ie=UTF8&psc=1) with active cooling (cooler on HIGH 5V pin). 

I made stress tests for the base 1.5GHz and overclocked to 2.15GHz with `over_voltage=6` and `arm_freq=2147`. I was initially a bit pessimistic about the case, but it looks like it did a pretty job in keeping the cpu below ~60-65°C. 

[image]

PS. I had to use the `--cooldown 10` for the overclocked case, as otherwise it wasn't getting within the default 0.2 °C range within 60 seconds (and also`-i 60` because I was rushing `:)`). Just in case I'm attaching the `.dat` files for both cases [data.zip](https://github.com/nschloe/stressberry/files/4094603/data.zip).

PPS. Thanks for the tool!

bobster316, 2021-02-26:
How did you modify the script to incrase max speed to 2.15Ghz?

haykh, 2021-02-26:
search "how to oveclock rpi 4". it's as simple as modifying a few flags in the boot txt file, but you gotta be careful with the parameters you put in.

closed 2021-02-26 by haykh