nschloe/stressberry #80: Vcgencmd Call

Alejandro-Roldan, 2020-07-25:
When running

```stressberry-run out.dat```

The error comes up

```
Awaiting stable baseline temperature...
Traceback (most recent call last):
  File "/usr/bin/stressberry-run", line 8, in <module>
    sys.exit(run())
  File "/usr/lib/python3.8/site-packages/stressberry/cli/run.py", line 92, in run
    cooldown(interval=args.cooldown, filename=args.temperature_file)
  File "/usr/lib/python3.8/site-packages/stressberry/main.py", line 14, in cooldown
    prev_tmp = measure_temp(filename=filename)
  File "/usr/lib/python3.8/site-packages/stressberry/main.py", line 37, in measure_temp
    out = subprocess.check_output(["vcgencmd", "measure_temp"]).decode("utf-8")
  File "/usr/lib/python3.8/subprocess.py", line 411, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.8/subprocess.py", line 489, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.8/subprocess.py", line 854, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.8/subprocess.py", line 1702, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'vcgencmd'
```

As we can see in here the problem is that it just calls vcgencmd

```
File "/usr/lib/python3.8/site-packages/stressberry/main.py", line 37, in measure_temp
    out = subprocess.check_output(["vcgencmd", "measure_temp"]).decode("utf-8")
```
So it would be better to change it to a full path

```out = subprocess.check_output(["/opt/vc/bin/vcgencmd", "measure_temp"]).decode("utf-8")```

After doing ```grep -rn vcgencmd /usr/lib/python3.8/site-packages/stressberry/*``` to see where else it needs to change

```/usr/lib/python3.8/site-packages/stressberry/main.py:51:        out = subprocess.check_output(["vcgencmd", "measure_clock arm"]).decode("utf-8")```

And do the same, change vcgencmd to the full path.

This apparently works. Will report back if I find any other issues.

**Done on a Raspberry Pi 4 with Archlinux ARM.**

nschloe, 2020-07-25:
Why is `/opt/vc/bin/` not in your path? Seems like a broken installation, right?

Alejandro-Roldan, 2020-07-26:
I never noticed that ```/opt``` wasn't in my path but I checked on my x86_64 Arch installation in my main computer and it doesn't have it either, furthermore it's able to use all the packages installed there because they have a symlink in ```/bin```; so I don't know about "broken installation". Maybe a distro thing?

nschloe, 2020-07-26:
Perhaps. It appears to work for everyone else.

Alejandro-Roldan, 2020-07-27:
Oh well, thanks you for your time :)

closed 2020-07-27 by Alejandro-Roldan

meafs, 2021-08-01:
I'm having the same issue on Ubuntu server 64 bit on my Raspberry Pi 4B.

Terminal:
```
ubuntu@ubuntu:~$ stressberry-run out.dat                                        Awaiting stable baseline temperature...
Traceback (most recent call last):
  File "/home/ubuntu/.local/bin/stressberry-run", line 8, in <module>
    sys.exit(run())
  File "/home/ubuntu/.local/lib/python3.8/site-packages/stressberry/cli/run.py", line 92, in run
    cooldown(interval=args.cooldown, filename=args.temperature_file)
  File "/home/ubuntu/.local/lib/python3.8/site-packages/stressberry/main.py", line 13, in cooldown
    prev_tmp = measure_temp(filename=filename)
  File "/home/ubuntu/.local/lib/python3.8/site-packages/stressberry/main.py", line 34, in measure_temp
    out = subprocess.check_output(["vcgencmd", "measure_temp"]).decode("utf-8")
  File "/usr/lib/python3.8/subprocess.py", line 415, in check_output
    return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
  File "/usr/lib/python3.8/subprocess.py", line 493, in run
    with Popen(*popenargs, **kwargs) as process:
  File "/usr/lib/python3.8/subprocess.py", line 858, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "/usr/lib/python3.8/subprocess.py", line 1704, in _execute_child
    raise child_exception_type(errno_num, err_msg, err_filename)
FileNotFoundError: [Errno 2] No such file or directory: 'vcgencmd'
```
Result of `uname -a` :
```
Linux ubuntu 5.4.0-1041-raspi #45-Ubuntu SMP PREEMPT Thu Jul 15 01:17:56 UTC 2021 aarch64 aarch64 aarch64 GNU/Linux
```

What can I do?