nschloe/stressberry #82: ERROR: stressberry-plot: error: unrecognized arguments: --hide-legend --line-width 2

Samk19, 2020-08-16:
Getting this Error stressberry-plot: error: unrecognized arguments: --hide-legend --line-width 2

When running command 
MPLBACKEND=Agg /home/pi/.local/bin/stressberry-plot mytest.out -f -d 300 -f -l 400 1600 -t 30 90 -o mytest.png --hide-legend --not-transparent --line-width 2

Works when remove —hide-legend and —line-width command but the graph is not clear

chunyianliew, 2021-01-17:
I am running in the same issue.

The two parameters have been removed in this [commit](https://github.com/nschloe/stressberry/commit/b02c5da84e0f622426f9df37fca347db880d73cf#diff-8b64539f58eef282dcf8b21979949a3bd8461c58a38be88c7bb375ac07472d8c) when introducing cleanplotlib.
And in the next [commit](https://github.com/nschloe/stressberry/commit/5ef2be41be61e8d21befb140b5cd6dc7ba840971#diff-8b64539f58eef282dcf8b21979949a3bd8461c58a38be88c7bb375ac07472d8c) cleanplotlib was replaced by dufte and in that commit the legend label plotting has become hardcoded.

For combined graphs that is not such a big issue, as it will only display the temperature y axis on the left. But when plotting the CPU frequency of one measurement together with the CPU frequency the temperature legend label overlaps the y axis showing the CPU frequency intervals.

[image]
[image]
[image]

shortbloke, 2021-06-09:
In my fork of this project I've reinstated `--hide-legend` and `--line-width` options https://github.com/shortbloke/stressberry

I've also added a `--legacy-style` to switch back to the previous type of graph and legend. Finally, I'm experimenting with adding a `--color-blind-friendly` option that changes the color pallet used to be more easily interpreted by those with colour blindness, though this is limited to use with `--legacy-style`.

More than happy to do a PR if @nschloe wants to allow the mixing of old and new styles.

p1r473, 2021-08-21:
@shortbloke please create a PR!

shortbloke, 2021-08-21:
@p1r473 Done. Not sure if it will be accepted.