nschloe/stressberry #90: need clarifications

Kyliael, 2020-12-16:
Hi,
I'm trying to get stressberry run on a Pi4 with a **fresh Raspbian OS install**.
I'm sure we need more details in the readme file

sudo apt install stress => OK
pip install stressberry => OK, no error to report

pi@raspberrypi:~ $ stressberry-run out.dat
-bash: stressberry-run: command not found

added /home/pi/.local/bin to PATH

export PATH=$PATH:/home/pi/.local/bin
echo $PATH
/home/pi/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games:/home/pi/.local/bin

pi@raspberrypi:~ $ stressberry-run out.dat
Traceback (most recent call last):
  File "/home/pi/.local/bin/stressberry-run", line 6, in <module>
    from stressberry.cli import run
  File "/home/pi/.local/lib/python2.7/site-packages/stressberry/__init__.py", line 15, in <module>
    from . import cli
  File "/home/pi/.local/lib/python2.7/site-packages/stressberry/cli.py", line 8, in <module>
    import matplotlib.pyplot as plt
  File "/home/pi/.local/lib/python2.7/site-packages/matplotlib/__init__.py", line 133, in <module>
    from matplotlib.rcsetup import defaultParams, validate_backend, cycler
  File "/home/pi/.local/lib/python2.7/site-packages/matplotlib/rcsetup.py", line 31, in <module>
    from matplotlib.fontconfig_pattern import parse_fontconfig_pattern
  File "/home/pi/.local/lib/python2.7/site-packages/matplotlib/fontconfig_pattern.py", line 28, in <module>
    from backports.functools_lru_cache import lru_cache
ImportError: No module named functools_lru_cache

pi@raspberrypi:~/.local/bin $ MPLBACKEND=Agg stressberry-run out.dat
Traceback (most recent call last):
  File "/home/pi/.local/bin/stressberry-run", line 6, in <module>
    from stressberry.cli import run
  File "/home/pi/.local/lib/python2.7/site-packages/stressberry/__init__.py", line 15, in <module>
    from . import cli
  File "/home/pi/.local/lib/python2.7/site-packages/stressberry/cli.py", line 8, in <module>
    import matplotlib.pyplot as plt
  File "/home/pi/.local/lib/python2.7/site-packages/matplotlib/__init__.py", line 133, in <module>
    from matplotlib.rcsetup import defaultParams, validate_backend, cycler
  File "/home/pi/.local/lib/python2.7/site-packages/matplotlib/rcsetup.py", line 31, in <module>
    from matplotlib.fontconfig_pattern import parse_fontconfig_pattern
  File "/home/pi/.local/lib/python2.7/site-packages/matplotlib/fontconfig_pattern.py", line 28, in <module>
    from backports.functools_lru_cache import lru_cache
ImportError: No module named functools_lru_cache

tried to follow many posts from this github like install libatlas-base-dev

pip3 install stressberry

stressberry-run out.dat
Awaiting stable baseline temperature...
Current temperature: 51.1°C - Previous temperature: 52.1°C

Seems to work...

If pip3 is the one to use, should be added to readme instead of pip
not sure about path, dependencies to install as well...

That would be great to have clear steps to follow starting from a **fresh official install.**

nschloe, 2020-12-16:
See https://github.com/nschloe/stressberry/issues/86.

Kyliael, 2020-12-16:
Well, still need more clarification
why pip installs stressberry but doesn't work
- why use pip3
- why readme doesn't state pip3

nschloe, 2020-12-16:
> why pip installs stressberry but doesn't work

Local config issue, see #86.

> why use pip3
> why readme doesn't state pip3

pip works the same way.

GODEXTREME, 2020-12-17:
> > why pip installs stressberry but doesn't work
> 
> Local config issue, see #86.
> 
> > why use pip3
> > why readme doesn't state pip3
> 
> pip works the same way.

pip installed the version 0.1.4 because it uses python 2.7...
I only fixed it using pip3, which installed the version 0.3.1 and uses python3

it's something like "it works on my machine in this way"

nschloe, 2020-12-17:
Sounds like a local config issue after all.

Kyliael, 2020-12-17:
Since pip doesn't worked as intended on a fresh official install (like Raspbian OS),  we should have detailed instructions to install it  in the Readme with pip3, etc..

nschloe, 2020-12-17:
So far, it wasn't a problem. If more people bump into this issue, I'll consider adding it to the readme.

GODEXTREME, 2020-12-17:
> So far, it wasn't a problem. If more people bump into this issue, I'll consider adding it to the readme.

pi@raspberrypi:~ $ pip install stressberry==0.3.1
DEPRECATION: Python 2.7 reached the end of its life on January 1st, 2020. Please upgrade your Python as Python 2.7 is no longer maintained. pip 21.0 will drop support for Python 2.7 in January 2021. More details about Python 2 support in pip can be found at https://pip.pypa.io/en/latest/development/release-process/#python-2-support pip 21.0 will remove support for this functionality.
Defaulting to user installation because normal site-packages is not writeable
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
ERROR: Could not find a version that satisfies the requirement stressberry==0.3.1 (from versions: 0.1.0, 0.1.1, 0.1.2, 0.1.3, 0.1.4, 0.1.5, 0.1.6, 0.1.7)
ERROR: No matching distribution found for stressberry==0.3.1

mezdahun, 2020-12-22:
I had the same issue on fresh Raspbian on Pi 4. Please add clarification to readme file. A single line to use python3 if multiple interpreters are present on the machine is enough but necessary...

FPSdoge, 2021-03-08:
same issue

Enji-Near, 2021-04-01:
I had the same issue. Please update the readme!

closed 2021-04-01 by nschloe