nschloe/stressberry #100: ERROR: Import error: no module named functools_lru_cache

mscatloaf, 2021-03-09:
I am running this program on a Raspberry pi to test overclocked performance. The program install fine, but when I run the command "MPLBACKEND=Agg stressberry-plot out.dat -o out.png" I recieve this error:

pi@raspberrypi:~ $ MPLBACKEND=Agg stressberry-plot out.dat -o out.png
Traceback (most recent call last):
  File "/home/pi/.local/bin/stressberry-plot", line 6, in <module>
    from stressberry.cli import plot
  File "/home/pi/.local/lib/python2.7/site-packages/stressberry/__init__.py", line 15, in <module>
    from . import cli
  File "/home/pi/.local/lib/python2.7/site-packages/stressberry/cli.py", line 8, in <module>
    import matplotlib.pyplot as plt
  File "/home/pi/.local/lib/python2.7/site-packages/matplotlib/__init__.py", line 133, in <module>
    from matplotlib.rcsetup import defaultParams, validate_backend, cycler
  File "/home/pi/.local/lib/python2.7/site-packages/matplotlib/rcsetup.py", line 31, in <module>
    from matplotlib.fontconfig_pattern import parse_fontconfig_pattern
  File "/home/pi/.local/lib/python2.7/site-packages/matplotlib/fontconfig_pattern.py", line 28, in <module>
    from backports.functools_lru_cache import lru_cache
ImportError: No module named functools_lru_cache

Note that I tried pip install functools which gave me this:

Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting functools
  Downloading https://files.pythonhosted.org/packages/22/3c/33589bf30422a92cf8c77054f2cf940ef2acc8a2857e5664045ed75a1c6a/functools-0.5.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/usr/lib/python2.7/dist-packages/setuptools/__init__.py", line 5, in <module>
        import functools
      File "functools.py", line 72, in <module>
        globals()['c_%s' % x] = globals()[x] = getattr(_functools, x)
    AttributeError: 'module' object has no attribute 'compose'
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-install-z8T8wu/functools/

I tried multiple variations of the same command which all had the same error at the end. This might be an issue with functools.

nschloe, 2021-03-09:
```
/home/pi/.local/lib/python2.7
```
Apparently this uses python 2.7. This version is more than 10 years old, and not supported anymore for a long time. No idea why it falls back to that on your system though.

mscatloaf, 2021-03-09:
I don't know, but I installed it normally. How could I run it with python3?

nschloe, 2021-03-09:
I think you can force it with
```
python3 `which stressberry`
```
but really your installation seems broken. There may be more errors down the line.

mscatloaf, 2021-03-09:
I did that, and it seems to have worked.

nschloe, 2021-03-09:
Alright then, cool!

closed 2021-03-09 by nschloe

mscatloaf, 2021-03-09:
Ok. using pip3 fixed it.