nschloe/stressberry #102: [--delta-t] and [infiles ...] . Error and some explanation please

akhilleusuggo, 2021-03-18:
```
$ stressberry-plot -h
usage: stressberry-plot [-h] [--version] [-o OUTFILE] [-t TEMP_LIMS TEMP_LIMS]
                        [-d DPI] [-f] [-l FREQ_LIMS FREQ_LIMS]
                        [--not-transparent] [--delta-t]
                        infiles [infiles ...]

```

Could you please explain the usefulness of this last 2 options ? And how to use them ?
Using `--delta-t` give an error .

```
 $ stressberry-plot test.dat -o test.png --delta-t
Traceback (most recent call last):
  File "/usr/local/bin/stressberry-plot", line 10, in <module>
    sys.exit(plot())
  File "/usr/local/lib/python3.7/dist-packages/stressberry/cli/plot.py", line 30, in plot
    plt.plot(d["time"], temperature_data, label=d["name"])
  File "/usr/local/lib/python3.7/dist-packages/matplotlib/pyplot.py", line 2842, in plot
    **({"data": data} if data is not None else {}), **kwargs)
  File "/usr/local/lib/python3.7/dist-packages/matplotlib/axes/_axes.py", line 1743, in plot
    lines = [*self._get_lines(*args, data=data, **kwargs)]
  File "/usr/local/lib/python3.7/dist-packages/matplotlib/axes/_base.py", line 273, in __call__
    yield from self._plot_args(this, kwargs)
  File "/usr/local/lib/python3.7/dist-packages/matplotlib/axes/_base.py", line 399, in _plot_args
    raise ValueError(f"x and y must have same first dimension, but "
ValueError: x and y must have same first dimension, but have shapes (156,) and (0,)
```

And if someone could explain this one , I would appreciate it .
```
positional arguments:
  infiles               input YAML file(s) (default: stdin)
```

shortbloke, 2021-06-09:
The `--delta-t` option is for if you have used stressberry run with the `--ambient` option and had a temperature sensor attached to the GPIO during the test measuring the ambient temperature. 
This allows for comparisons between test runs and test scenarios when the ambient temperature is inconsistent. (e.g. at my desk).

`infiles` is just the last argument you pass to `stressberry-plot` which is the name of the result file or a list of files you want to plot.
Hope that helps.

akhilleusuggo, 2021-06-10:
Thank you Sr!

closed 2021-06-10 by akhilleusuggo