nschloe/stressberry #107: steps for manual install (ie without pip)

graysky2, 2021-06-24:
Can you detail some steps to install without using pip?

nschloe, 2021-06-24:
Not possible, I don't think.

graysky2, 2021-06-24:
Install dependencies, then...?

nschloe, 2021-06-24:
Then what?

graysky2, 2021-06-24:
That's what I'm asking :D

The logs from pip:
```
% pip install stressberry
Defaulting to user installation because normal site-packages is not writeable
Collecting stressberry
  Downloading stressberry-0.3.3-py3-none-any.whl (23 kB)
Collecting matplotlib
  Downloading matplotlib-3.4.2-cp39-cp39-manylinux2014_aarch64.whl (9.0 MB)
     |████████████████████████████████| 9.0 MB 57 kB/s 
Collecting dufte
  Downloading dufte-0.2.20-py3-none-any.whl (18 kB)
Collecting pyyaml
  Downloading PyYAML-5.4.1-cp39-cp39-manylinux2014_aarch64.whl (788 kB)
     |████████████████████████████████| 788 kB 11.1 MB/s 
Collecting numpy
  Downloading numpy-1.21.0-cp39-cp39-manylinux_2_17_aarch64.manylinux2014_aarch64.whl (13.0 MB)
     |████████████████████████████████| 13.0 MB 8.3 MB/s 
Requirement already satisfied: pyparsing>=2.2.1 in /usr/lib/python3.9/site-packages (from matplotlib->stressberry) (2.4.7)
Collecting python-dateutil>=2.7
  Downloading python_dateutil-2.8.1-py2.py3-none-any.whl (227 kB)
     |████████████████████████████████| 227 kB 13.3 MB/s 
Collecting cycler>=0.10
  Downloading cycler-0.10.0-py2.py3-none-any.whl (6.5 kB)
Requirement already satisfied: pillow>=6.2.0 in /usr/lib/python3.9/site-packages (from matplotlib->stressberry) (8.2.0)
Collecting kiwisolver>=1.0.1
  Downloading kiwisolver-1.3.1-cp39-cp39-manylinux2014_aarch64.whl (78 kB)
     |████████████████████████████████| 78 kB 2.3 MB/s 
Requirement already satisfied: six in /usr/lib/python3.9/site-packages (from cycler>=0.10->matplotlib->stressberry) (1.16.0)
Installing collected packages: python-dateutil, numpy, kiwisolver, cycler, matplotlib, pyyaml, dufte, stressberry
  WARNING: The scripts f2py, f2py3 and f2py3.9 are installed in '/home/facade/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
  WARNING: The scripts stressberry-plot and stressberry-run are installed in '/home/facade/.local/bin' which is not on PATH.
  Consider adding this directory to PATH or, if you prefer to suppress this warning, use --no-warn-script-location.
Successfully installed cycler-0.10.0 dufte-0.2.20 kiwisolver-1.3.1 matplotlib-3.4.2 numpy-1.21.0 python-dateutil-2.8.1 pyyaml-5.4.1 stressberry-0.3.3
```

So I'm wondering what I can manually do to not use pip (ie for packaging).

nschloe, 2021-06-24:
> So I'm wondering what I can manually do to not use pip (ie for packaging).

Like I said, it's not possible to install stressberry without pip.

closed 2021-06-24 by nschloe

graysky2, 2021-06-25:
> Like I said, it's not possible to install stressberry without pip.

It is, users of Arch ARM can use the following two PKGBUILDs to access this software.  Might want to update your readme?

https://aur.archlinux.org/packages/python-dufte/
https://aur.archlinux.org/packages/stressberry/

graysky2, 2021-06-25:
Also added to the official Arch ARM repos: https://github.com/archlinuxarm/PKGBUILDs/commit/cd7f495939cba6770818f54b57e2994abe7569a9#diff-53a6b6de3ececa1f8d0edf4ebc9265719467bcf3d23be06dde80d27c09cac31f