nschloe/stressberry #112: ImportError: with stressberry . On the raspberry pi 4

ChrisAFRcrew, 2021-09-02:
I have followed all the steps and now hitting a Error. Which I don't know how to fix. Everything is working just the plot is not. When I type in the terminal of the raspberry pi 4, I get an error message. "ImportError: bumpy.core.multiarray failed to import." I type it two ways, I typed "stressberry-plot out.dat -o out.png" , which I received the error. Than I typed, "MPLBACKEND=Agg stressberry-plot out.dat -o out.png" and still get the error message.  Do have to take other steps. If so please let me know. Thank you

embeddedaddicted, 2021-11-02:
Hello all,

for me this works on Pi Zero 2 W....

sudo apt-get install libatlas-base-dev libopenjp2-7-dev libtiff5

Best regards
Marc

ChrisAFRcrew, 2021-11-05:
Thank and do apologize for the late response

YaiGenes, 2022-06-11:
Hi, you just need to install numphy in your console. Follow this command:
`pip install -U numpy`

For more details follow this conversation:

[https://stackoverflow.com/questions/20518632/importerror-numpy-core-multiarray-failed-to-import](importerror-numpy-core-multiarray-failed-to-import])