nschloe/stressberry #118: Can you update for deprecation of python-dufte

graysky2, 2021-12-03:
I see [dufte](https://github.com/nschloe/dufte#deprecation) is deprecated and replaced.  Can you update stressberry to work with the replacement?  When I try:
```
% stressberry-run -n "comments" fan.dat                                                           
Awaiting stable baseline temperature...
...
% stressberry-plot -o caseless.png -f -l 200 2200 -t 20 90 fan.dat 
Traceback (most recent call last):
  File "/usr/bin/stressberry-plot", line 8, in <module>
    sys.exit(plot())
  File "/usr/lib/python3.9/site-packages/stressberry/cli/plot.py", line 9, in plot
    import dufte
ModuleNotFoundError: No module named 'dufte'
```

Upstream or am I doing something wrong?

graysky2, 2021-12-03:
Ah, is it as simple are removing line 9 from `plot.py`?

https://github.com/nschloe/stressberry/blob/main/stressberry/cli/plot.py#L9

nschloe, 2021-12-03:
Deprecation is just a warning, I'll get to it at some point.

graysky2, 2021-12-03:
Ah, is it as simple are removing line 9 from plot.py?

https://github.com/nschloe/stressberry/blob/main/stressberry/cli/plot.py#L9

commit 55501513 (2021-12-04): convert to matplotx from dufte, fixes #118
commit b3a8bc46 (2021-12-04): Merge pull request #119 from graysky2/new

convert to matplotx from dufte, fixes #118
closed 2021-12-04 by nschloe