nschloe/stressberry #123: Raspberry Pi 4B with default case

AstroOrbis, 2022-02-07:
Running ubuntu server  and a few background tasks with the default case, connected to ethernet and plugged into power 24/7
[image]

mattwelke, 2022-07-30:
I have a similar setup. Ubuntu server, stock case, stock clock. Funny, my graph ended up very similar to yours. A few degrees lower, maybe lower ambient temperature. Or maybe due to the 3 small aluminum heat sinks I used on it.

[image]

AstroOrbis, 2022-07-30:
Nice graph! Where'd you get the heatsinks? I've changed my setup since then, now I have a small stack running with a fan pointed directly at it (part of a kit I bought from Micro Center), and I'd love to upgrade it with those.

mattwelke, 2022-07-30:
I used to get my Pis from Canakit. Their kits always included a few aluminum heatsinks, one for the CPU, one for RAM, and one for USB. Example: https://www.amazon.ca/CanaKit-Raspberry-Pi-Extreme-Kit/dp/B09Q4TQBSZ

[image]

I haven't done testing comparing temps with and without the heatsinks. Today I just switched from a Canakit case to a Flirc case, which is a metallic, passive heatsink style case: https://www.amazon.ca/gp/product/B07WG4DW52

And I can say for sure that it helped a lot. Brought my temps down from 73 C idle to about 48 C idle. I was able to overclock to 2 GHz for turboing and still keep it under 70 C when it's pinned at 2 GHz under load.