nschloe/stressberry #129: Arch Linux doesn't appear to be increasing CPU activity

s-fairchild, 2022-08-23:
I've installed stressberry via `pacman`. Running `stressberry-run` doesn't increase the CPU utilization at all.
I haven't seen it go above 1% and it appears to only have 2 additional processes running even though I specified 4.
I've tried running as a standard user first, then root, then adjusting the nice value (although that really doesn't matter).

Here's my tests
```bash
stressberry-run -c 4 stressberry_ICETOWER_PWMFAN_$(date +%F)-test2.txt
sudo nice -n -10 stressberry-run -c 4 -n 'RPI4 IceTower Fan 1 hour test' -d 60 "stressberry_ICETOWER_PWMFAN_$(date +%F_%H:%M)-test2.txt"
```

Am I miss understanding how stressberry works?

[image]

graysky2, 2022-11-11:
Works for me: https://archlinuxarm.org/forum/viewtopic.php?f=3&t=15400

AlexBaranowski, 2023-01-23:
Can you check if `stress` is available? I had similar problem with `stress-ng` not being symlinked to the `stress`.

s-fairchild, 2023-02-07:
@AlexBaranowski Hi Alex, yes stress is installed.
```bash
[steven@rpi4:~ $]$ stress --version
stress 1.0.6
```

millerii, 2023-06-28:
> Can you check if `stress` is available? I had similar problem with `stress-ng` not being symlinked to the `stress`.

Worked for me by installing both 'stress' (already installed) and also 'stress-ng'