nschloe/stressberry #135: error: externally-managed-environment

geekworm-com, 2023-11-16:
I found the following errors on Raspberry Pi 5 Model B;

```
pi@raspberrypi:~/stressberry $ python3 -m pip install stressberry
error: externally-managed-environment

This environment is externally managed

To install Python packages system-wide, try apt install
python3-xyz, where xyz is the package you are trying to
install.

If you wish to install a non-Debian-packaged Python package,
create a virtual environment using python3 -m venv path/to/venv.
Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
sure you have python3-full installed.

For more information visit http://rptl.io/venv

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
```

Are there some updates?

shortbloke, 2023-11-17:
I believe this is expected with Python 3.11 and the implementation of [PEP668](https://peps.python.org/pep-0668/) as per the message you have two options in this case:
1. Install with the flag `--break-system-packages`
2. setup a venv and run pip and stressberry from there. 

I believe if you installed Python  3.11 or later on any RPi 4 you'd see the same issue.

Ate329, 2024-01-06:
I encountered the same issue and you can try to use this command (if you are using python3.11): 
```sudo mv /usr/lib/python3.11/EXTERNALLY-MANAGED /usr/lib/python3.11/EXTERNALLY-MANAGED.old```
There are other solutions in this stackoverflow question: https://stackoverflow.com/questions/75608323/how-do-i-solve-error-externally-managed-environment-every-time-i-use-pip-3