nschloe/stressberry #137: Test case : Pi 5 B with aluminium foil case

ChingWeiChan, 2024-03-11:
Thank you for the packages. I try three condition:
* without case
* with case but do not startup fans (do not connect the power)
* with case and activate the fans

It seems we have a good effect in this case whether fans are activated.
[image]

It's the case picture.
[image]

banitt, 2025-03-10:
How did you managed to get this image? I am getting something different - no grid, different labels:

[image]

ChingWeiChan, 2025-03-10:
Hi @banitt , I use the output `out.dat` data and convert it to csv file. And use my self-writen python script with `matplotlib` to print the image.